WildernessLabs/Documentation #45: Updating firmware page is a bit hard to follow

chris1seto, 2017-08-05:
"Download the latest Netduino firmware files from the downloads page. You’ll need the Tinybooter, ER_CONFIG, ER_FLASH files for your specific Netduino model."

What package are these files inside? I see the Netduino3Wifi_4_3_2_2 dir, but the files are the wrong extension for MFDeploy

bryancostanich, 2017-08-07:
Note to self: files are .s19, which is fine for the Netduino windows updater and the Mac Deploy, but MFDeploy expects .hex.

bryancostanich, 2017-09-30:
Further note to self: need to remove the MFDeploy stuff and replace with the Secret Labs Netduino updater, since it's better than MFDeploy; it'll unbrick a device, or deploy to a device that doesn't have tinybooter because it uses DFU, rather than MFUSB.

commit e1ade9af (2018-01-25): Updated firmware docs to use netduino updater. fixes #45.
closed 2018-01-25 by bryancostanich