WildernessLabs/Documentation #66: Add linux installation instructions

bryancostanich, 2018-01-25:
need some help here.

Create a udev rule at /etc/udev/rules.d/50-netduino.rules with the content SUBSYSTEMS=="usb", ATTR{idProduct}=="1000", ATTR{idVendor}=="22b1", MODE="0660", GROUP="users", TAG+="uaccess" (this makes the Netduino accessible to standard users).

dgr1992, 2020-12-15:
Hi,
I finally found some time to work with my Meadow F7 micro. I managed to do the update of the Meadow OS using Ubuntu and also created a Pulse LED program using the dotnet CLI. Unfortunately I am now getting a System.Object not found error when running the app. Can you give ma hint what to check?

Documentation wise, where can I add the installation instruction?

bryancostanich, 2020-12-17:
@dgr1992 can you join the public slack so we can troubleshoot, please? http://slackinvite.wildernesslabs.co/

the linux setup docs would likely go here: https://github.com/WildernessLabs/Documentation/blob/master/docs/Meadow/Getting_Started/Deploying_Meadow/index.md

closed 2021-01-01 by bryancostanich