WildernessLabs/Documentation #78: Home page images should appear under the heading in tablet/phone responsive sizes

bryancostanich, 2018-01-28:
Right now, they're winding up under the bottom of the content:

[image]

stephenkarpeles, 2018-01-29:
Images appearing under headings on mobile/tablet.

bryancostanich, 2018-01-29:
Pulled latest on the stephenkarpeles branch and still seeing this:

[image]

stephenkarpeles, 2018-02-04:
Just pushed another fix for this...

bryancostanich, 2018-02-04:
Looks good, except for the Netduino.Foundation one:

[image]

stephenkarpeles, 2018-02-05:
Okay. Should be updated.

closed 2018-02-05 by bryancostanich