WildernessLabs/Documentation #88: Add Hackster Projects to next step

bryancostanich, 2018-06-12:
we need to point to the hackster projects at the end of our getting started doc
so it goes: *Netduino Getting Started* > *Netduino.Foundation Getting Started* > *Hackster Projects*
first two are done:
* http://developer.wildernesslabs.co/Netduino/Getting_Started/
* http://netduino.foundation/Getting_Started/
but that then sends folks to the peripheral library
we should send to hackster
and on the netduino getting started docs, we should have a list at top that enumerates those steps

adrianstevens, 2018-06-29:
list of published huckster projects can be found on Bryan's or Jorge's profile: https://www.hackster.io/bryan-costanich

jorgedevs, 2021-01-01:
hey @bryancostanich, revisiting this issue, I noticed that the Netduino.Foundation website is down. 

http://netduino.foundation/Getting_Started/

Is this expected?

bryancostanich, 2021-01-01:
weird. the whole site is not right: http://netduino.foundation/

patridge, 2022-11-04:
fwiw, the site issue appears to be reported in #239

patridge, 2022-11-05:
For the initial request, since we can't access netduino.foundation, we could so a table of projects under the **Next - ...** section, kind of like we do on Meadow pages. Would this work?

[image]

If that is a good workaround for this, here's the PR awaiting being published from draft format.
https://github.com/WildernessLabs/Documentation/pull/380

patridge, 2023-01-25:
That PR to add the Hackster projects to our current docs site is merged. The old netduino.foundation site being down is still an issue. And if we want this list on that site as well, we can keep this issue rolling as well.