WildernessLabs/Documentation #101: Add Hack Kit Page

bryancostanich, 2019-04-11:
We need a developer.wildernesslabs.co/HackKit page that has the hack kit contents and links to projects and whatnot you can build with it.

To start with, just having the page with the contents of the Hack Kit Pro and Lite enumerated would be good. A number of beta folks have asked what's all in the Hack Kit. So we should publish the list. The items can be pulled from [here](https://docs.google.com/spreadsheets/d/12sGftx6Xi3KyIgBwYzxhQUVuXv-_pHFONNE32ABmrPw/edit#gid=1184041558)

patridge, 2019-09-11:
Thanks to @lamebrain, I found this page with the HackKit details. It doesn't seem to be redirected from `/HackKit` on the non-beta subdomain, though.
http://beta-developer.wildernesslabs.co/HackKit/

jorgedevs, 2021-01-01:
Closing this issue, since we in fact have a [Hack Kit Page](http://developer.wildernesslabs.co/HackKit/) for sometime now with the components included and links to hackster projects.

closed 2021-01-01 by jorgedevs