WildernessLabs/Documentation #112: SPI Bug & Documentation Issue

GeorgeStephens, 2019-10-02:
There seems to be an issue with the SPI library that while works in some cases if you follow the [SPI](http://beta-developer.wildernesslabs.co/Meadow/Meadow_Basics/IO/Digital/Protocols/SPI/) documentation you will receive an exception. 

When used as in the [WaveShare_ePaper_Sample](https://github.com/WildernessLabs/Meadow.Foundation/blob/master/Source/Samples/Displays/WaveShare_ePaper_Sample/WaveShareEPaperApp.cs) everything is fine:
`ISpiBus SPI = Device.CreateSpiBus();`

But using it as the documentation shows through's an exception:
`ISpiBus SPI = Device.CreateSpiBus(Device.Pins.Groups.Spi1);` stating that 

> [0x4:] EXCEPTION handling: System.NullReferenceException: Object
>  reference not set to an instance of an object.

If you go through the library/intelisense and use `ISpiBus SPI = Device.CreateSpiBus(Device.Pins.SCK, Device.Pins.MOSI, Device.Pins.MISO);` the application just hangs forever (okay well more than 10 minutes as thats all i gave it...) with no exception being generated.

May I suggest that the docs be changed to use `ISpiBus SPI = Device.CreateSpiBus();` until bug is fixed? Unless it's going to be fixed soon but then with the current Meadow board there is only the one usable port anyway so maybe it should be changed to that permanently for simplicity?

bryancostanich, 2019-10-04:
Thanks, George! We're working on the underlying issue for beta 3.4. In the meantime, docs are fixed.

closed 2019-10-04 by bryancostanich

GeorgeStephens, 2019-10-24:
Worth noting that changing the bus speed doesn't seem to be respected either...

Whether you leave it at its default (1Mhz?) or set it to another value (say `200` for 200 KHz) the output is always around 375KHz.

[image]