WildernessLabs/Documentation #115: bootloader driver issue

faaland, 2019-10-12:
I could not find a troubleshooting page, but you may want to add that for Windows, dfu-util may report "Cannot open DFU device 0483:df11"

I had to use zadig (https://zadig.akeo.ie/) to install WinUSB instead of whatever it had already.

Reference:  https://sourceforge.net/p/dfu-util/tickets/75/

Vliedberger, 2019-10-12:
I had the same issue, thanks for the hint of using Zadig! Only after replacing the usb driver I was able to load Meadow OS on my Windows 10 machine.
[image]

mtabet, 2019-10-12:
Thanks Guys, that worked but is there a Meadow slack channel or forum or something to look up latest issues? Hitting one now in Visual Studio....

NevynUK, 2019-10-13:
Issues with Meadow can be reported in the [Meadow Issues](https://github.com/WildernessLabs/Meadow_Issues) repository.

I think you can find the public Slack channels at [wildernesslabspublic.slack.com](wildernesslabspublic.slack.com) or search in the Slack application for _Wilderness Labs External_.

Regards,
Mark

mtabet, 2019-10-13:
http://slackinvite.wildernesslabs.co

closed 2021-01-01 by bryancostanich