WildernessLabs/Documentation #136: The instructions can lead you into an issue with installing standoffs on the wooden platform

mlinnen, 2019-11-16:
I would suggest optionally installing the standoffs first on the wooden platform before you mount the breadboard and meadow.  Doing it the other way you might not have enough room to put the standoffs on last.

patridge, 2021-07-21:
I just went through this process for a new Meadow board without any standoffs available. (They aren't included in any of the prior kits either.)
Since standoffs are an optional step added at the bottom, it could work to mention not covering the corner holes when mounting the breadboard if folks wish to use those holes for that purpose later.

patridge, 2024-06-24:
Page: https://developer.wildernesslabs.co/Meadow/Getting_Started/Assemble_Meadow/
Content: https://github.com/WildernessLabs/Documentation/blob/develop/docs/Meadow/Getting_Started/Assemble_Meadow/index.md

PR #778 (2024-06-24): Add note on hole clearance
{fixes #136}
closed 2024-06-24 by patridge