WildernessLabs/Documentation #138: Cannot deploy to meadow anymore

Norbert62, 2019-11-19:
After playing a little bit with my new meadow board and trying to connect a tft display I suddenly am not longer able to deploy my application to the board. This is the log of the VS output window:
1>------ Deploy started: Project: MeadowTest03System, Configuration: Debug Any CPU ------
1>Deploying to Meadow on COM3...
1>Initializing Meadow
1>Checking files on device (may take several seconds)
1>Writing App.exe
1>Das Zeitlimit für die Semaphore wurde erreicht.
1>
========== Build: 0 succeeded, 0 failed, 1 up-to-date, 0 skipped ==========
========== Deploy: 0 succeeded, 1 failed, 0 skipped ==========

I already flashed the newest version of Meadow.OS and Meadow.Runtime with no success. Is there a way to erase the entire device and start over?

csuski, 2019-11-27:
@Norbert62  When I got this error the only way I could recover was to use the Meadow.CLI program http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/ program with the --FormatFileSystem argument. After that I was able to reload a program (although it took a really long time).

Norbert62, 2019-12-06:
Yes, thanks, that worked. However, now deploying apps to the board is very slow. And still I am not able to deploy an app I wrote to to the board (always getting timeouts).

csuski, 2019-12-06:
It does take a long time the first time. Try deploying the sample app that just blinks the onboard. Future deploys will be much faster.

However, I found that the program I deployed that caused the issue continued to cause the issue and it was specific dlls. I was deploying:

https://github.com/WildernessLabs/Meadow_Issues/issues/46

closed 2021-01-01 by bryancostanich