WildernessLabs/Documentation #141: Documentation links are incorrect

aholmes, 2019-11-25:
After the reorg commit https://github.com/WildernessLabs/Meadow.Foundation/commit/7e80d1680edf584eda5a8b981edc12becce9e3bb, some documentation links are now invalid. For example:

http://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Displays.Lcd.CharacterDisplay.html#Meadow_Foundation_Displays_Lcd_CharacterDisplay_WriteLine_System_String_System_Byte_
The View Source link for this method links to https://github.com/WildernessLabs/Meadow.Foundation/blob/origin/master/Source/Meadow.Foundation.Peripherals/Displays.Lcd.CharacterDisplay/Driver/Displays.LCD.CharacterDisplay/CharacterDisplay.cs/#L154

The "origin" part of the URL is invalid.

I suggest linking to either just "master" https://github.com/WildernessLabs/Meadow.Foundation/blob/master/Source/Meadow.Foundation.Peripherals/Displays.Lcd.CharacterDisplay/Driver/Displays.LCD.CharacterDisplay/CharacterDisplay.cs

or to the specific SHA https://github.com/WildernessLabs/Meadow.Foundation/blob/7e80d1680edf584eda5a8b981edc12becce9e3bb/Source/Meadow.Foundation.Peripherals/Displays.Lcd.CharacterDisplay/Driver/Displays.LCD.CharacterDisplay/CharacterDisplay.cs

CartBlanche, 2019-11-25:
Here's a partial PR (I only fixed the pages I was looking at a few days ago) - https://github.com/WildernessLabs/Documentation/pull/139. More work is required.

jorgedevs, 2021-01-01:
Hello @aholmes, this issue has been solved at some point during several releases, but feel free to re-open them if you find any other borked links.

closed 2021-01-01 by jorgedevs