WildernessLabs/Documentation #150: Is dfu-util v0.8 (32-bit) supported?

ellisnet, 2019-12-16:
The current version of dfu-util for Windows is v0.9 - and this is the version that is linked to from the Meadow OS Deployment page.  However, this version (0.9) appears to be 64-Bit Windows only.  If a person instead downloads the previous version (0.8), that appears to work fine on 32-Bit Windows; i.e. the switch from 32-Bit to 64-Bit occurred between version 0.8 and 0.9 (current version) of dfu-util.
So, is version 0.8 supported for deploying the Meadow OS to boards on 32-Bit Windows devices? Or do you have to have 0.9 of dfu-util (and therefore 64-Bit Windows)?
Can the documentation be updated to clarify this situation?

ellisnet, 2019-12-16:
(In case anyone is curious, I was wondering about this because the new Surface Pro X from Microsoft is running Windows on ARM.  This is 64-Bit Windows on ARM; but it is only compatible with x86 - 32-bit Intel - applications, and cannot run x64 - 64-bit Intel - applications.  I confirmed that v0.8 of dfu-util *does* run on the Surface Pro X; and v0.9 of dfu-util does not.  But then I got into the Zadig issue; and decided not to mess with USB drivers on my Surface Pro X.)

bryancostanich, 2021-01-01:
this flow has changed significantly from when this was filed.

closed 2021-01-01 by bryancostanich