WildernessLabs/Documentation #198: Bridge rectifier diagram incorrect

colinturner61, 2020-05-30:
In the document on general diodes http://developer.wildernesslabs.co/Hardware/Tutorials/Electronics/Part6/General_Diodes/ , the corners of the bridge are connected to the wrong current type. The AC input should only have one path it can take in each of its +ve and -ve phases.

bryancostanich, 2021-01-01:
I'm not sure what you mean. In a full-bridge rectifier, as shown, both common and hot get connected to the bridge, which allows it to provide a positive voltage on both cycles, forward and back. See here: https://www.electronics-tutorials.ws/diode/diode_6.html

colinturner61, 2021-01-04:
In the diagram here: http://developer.wildernesslabs.co/Hardware/Tutorials/Electronics/Part6/General_Diodes/ the direction of the bottom two diodes should be reversed.
For example on the positive part of the sine wave, the current should only be able to flow through the top left diode. It should then pass through the load and then back through the bottom right diode. However, as the bottom two diodes are each reversed, the positive ac current could split beween both left hand diodes and then no return flow from the load is possible.

bryancostanich, 2021-01-04:
heh, you're totally right! I misunderstood your initial post!

PR #381 (2022-11-05): Rotate two misaligned diodes
Fixes #198 

I did a quick adjustment to the diode element's `rotate` transformation operations to manually flip things.
closed 2022-11-05 by adrianstevens