WildernessLabs/Documentation #208: Links broken on b4.0 annoncements

mongeon, 2020-09-22:
Under the Meadow.Foundation portion of the b4.0 announcments, the 3 links are 404.

> Meadow.Foundation has new peripheral drivers to play with, including:
> - Mpr121
> - Vl53l0x
> - Vc0706

Broken urls:

- http://beta-developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Sensors.Hid.Mpr121.html

- http://beta-developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Sensors.Distance.Vl53l0x.html

- http://beta-developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Sensors.Camera.Vc0706.html

jorgedevs, 2021-01-01:
Thanks for reporting, the links should be working now. Also corrected that the Camera driver is not live in the NuGet gallery, as it needs serial events which we don't have them working yet.

closed 2021-01-01 by jorgedevs

mongeon, 2021-01-01:
@jorgedevs The links are still broken on my side, I've made sure to refresh the cache but still got the 404 error code.

jorgedevs, 2021-01-01:
Hello @mongeon, I noticed you're referring to our Beta site. It has been fixed it on the production site: http://developer.wildernesslabs.co/Meadow/Release_Notes/Beta4/

Beta site is a bit behind with some of our latest changes, so I suggest you go to our production site instead.

Cheers,
Jorge