WildernessLabs/Documentation #237: Purchase links are generally out of date

adrianstevens, 2021-01-17:
Ideas:
- review and update
- remove purchase section

And/or 
- add a general purchasing page

patridge, 2022-11-04:
@adrianstevens Got any more details on this? There is this general store page, if that's what was needed: https://store.wildernesslabs.co/.

adrianstevens, 2024-06-24:
@adrianstevens sorry we had purchase links on some of the peripheral pages that linked to peripherals on Amazon, AliExpress, etc.

Those links go bad overtime so I recommend we delete (if any still exist)

patridge, 2024-06-24:
Some quick searches:
* ["amazon"](https://github.com/search?q=repo%3AWildernessLabs%2FDocumentation%20amazon&type=code)
* ["ebay"](https://github.com/search?q=repo%3AWildernessLabs%2FDocumentation+ebay&type=code)
* ["ali"](https://github.com/search?q=repo%3AWildernessLabs%2FDocumentation+aliexpress&type=code)

adrianstevens, 2024-06-24:
I suspect Adafruit will be in there somewhere too

patridge, 2024-06-24:
Updated list with adafruit.com and sparkfun.com (".com", since they have their name on all sorts of products directly).

patridge, 2024-06-24:
That said, if we are describing a particular product that they make, maybe those links should stay and be updated. 🤷

adrianstevens, 2024-06-24:
I vote - no links to stores, unless you're volunteering to validate links and update at least weekly