WildernessLabs/Documentation #245: Exceptions thrown when disabling or enabling Mono

bmitc, 2021-02-21:
When performing steps 3 and 4 in [this documentation](http://developer.wildernesslabs.co/Meadow/Getting_Started/Deploying_Meadow/), exceptions are thrown. However, it does appear the commands work correctly because the other steps seem to complete okay.

```
PS C:\Users\bmitc> meadow --MonoDisable -s COM6
Opening port 'COM6'
Data: Mono disabled, will not run app.exe
Data: Mono has been disabled. Restarting Meadow
Unexpected error occurred: Exception of type 'MeadowCLI.DeviceManagement.MeadowDeviceManagerException' was thrown.
```

```
PS C:\Users\bmitc> meadow --MonoEnable
Opening port 'COM6'
Data: Mono has been enabled. Restarting F7 Micro
Unexpected error occurred: Exception of type 'MeadowCLI.DeviceManagement.MeadowDeviceManagerException' was thrown.
```

bmitc, 2021-02-21:
I am getting the same exception when trying to deploy the F# hello world application to Meadow, so now I think there's probably some error in the serial communication.

For example, deploying the F# application yields the following output:

```
Build started...
1>------ Build started: Project: MeadowApplication1, Configuration: Debug Any CPU ------
1>MeadowApplication1 -> C:\Users\bmitc\source\repos\MeadowApplication1\MeadowApplication1\bin\Debug\net472\App.exe
2>------ Deploy started: Project: MeadowApplication1, Configuration: Debug Any CPU ------
2>[2/21/2021 3:02:27 PM]    Deploying to Meadow on COM6...
2>Exception of type 'MeadowCLI.DeviceManagement.MeadowDeviceManagerException' was thrown.
========== Build: 1 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
========== Deploy: 0 succeeded, 1 failed, 0 skipped ==========
```

bmitc, 2021-03-08:
I was wondering if there are any thoughts on this. This issue actually probably belongs in one of the other repositories, but I am unable to move it.

I am unable to run the [F# template application](https://github.com/WildernessLabs/Meadow.Sdk/tree/main/Meadow_DotNet_SDK/Project_Templates/templates/MeadowApplicationFSharp) on my board due to this error.

bmitc, 2022-01-12:
Are there any updates on this issue?

patridge, 2022-11-04:
This appears to be resolved at some point. I am able to do a file-new F# app in VS2022 and deploy fine without any exceptions around the Mono enable/disable steps.
If you are still seeing this issue, definitely drop in any new info.

closed 2022-11-04 by patridge

bmitc, 2022-11-05:
Thanks for following up! I haven't used my Meadow since then due to this bug, so maybe I'll get it out again if it's not happening now.

patridge, 2022-11-05:
Definitely worth another round! The APIs are getting really smooth and enjoyable to use. And with JIT enabled, things got significantly faster.

This was my first time doing a Meadow app in F#, which was quite fun. Definitely share if you see any other issues.