WildernessLabs/Documentation #253: Need to update Meadow_Basics/IO/Digital

adrianstevens, 2021-03-22:
http://developer.wildernesslabs.co/Meadow/Meadow_Basics/IO/Digital/

Review and update - especially the bottom of the page

NevynUK, 2023-10-02:
Just trying to get a user application working and some compiler errors cropped up so wandered off to the documentation and I think the samples and the documentation are out of step.

Sample:
https://github.com/WildernessLabs/Meadow.Core.Samples/blob/main/Source/IO/DigitalInputPort/MeadowApp.cs

Documentation:
http://developer.wildernesslabs.co/Meadow/Meadow_Basics/IO/Digital/

And taking code from the sample does not seem to compile with the latest system.

patridge, 2024-06-24:
@NevynUK Do you know what was/is out of sync still, any specific code? I think some of this page was updated since last October. I'm hoping it's in a happy place now.