WildernessLabs/Documentation #254: Timer groups/channels are undocumented for PWM

bryancostanich, 2021-03-24:
PWMs are created on timer groups and channels. multiple pins share the same group but are on different channels. 

all the PWMs on a single timer must share the same frequency, but can have different duty cycles. 

today, Meadow.Core enforces this and will throw an exception if you try to set competing frequencies on the same timer group, but no where in docs do we talk about the timer groups.

for source inspiration, we do document the interrupt groups.

we'll need to modify the pinout diagram, as well as add docs to describe.

jorgedevs, 2021-04-05:
Done and available in the [PWM docs](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/IO/Digital/PWM/)

closed 2021-04-05 by jorgedevs

bryancostanich, 2021-04-05:
Awesome work, thanks, @jorgedevs !