WildernessLabs/Documentation #255: Meadow's RTC is undocumented

bryancostanich, 2021-03-26:
We have this fantastic [hackster example](https://www.hackster.io/wilderness-labs/build-a-clock-with-meadow-s-onboard-real-time-clock-chip-2b1f85) Jorge did, but we don't have any docs in the docs site on the RTC, except for a note about it in the [power](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/IO/Power/) guide. 

we should add a page under /Meadow.OS/ that talks about the RTC, the API, and such.

patridge, 2022-11-05:
@bryancostanich or @jorgedevs, does the current Clock page work for this, or were you thinking something else for this?

http://developer.wildernesslabs.co/Meadow/Meadow.OS/RTC/

bcr, 2023-04-08:
The instructions there use `Device.SetClock` which I don't think exists anymore and you need to use `Device.PlatformOS.SetClock`

patridge, 2023-04-12:
Can confirm that in the latest release VS2022 extension. It looks like that `Device.PlatformOS.SetClock` is the correct method going forward.

That said, it may be easier to set the clock via NTP at startup, like it appear to be doing in this [countdown clock Hackster project](https://www.hackster.io/wilderness-labs/wifi-christmas-countdown-timer-w-an-lcd-display-and-meadow-e4cf9c).

I'll get that docs page updated regardless.

patridge, 2023-10-02:
Fixed

PR #441 (2023-04-12): Update SetClock to new namespace
Fixes #255 in docs, but still requires changes to Hackster.

@jorgedevs Can you update the `Device.SetClock` uses here to align with the updated [sample source](https://github.com/WildernessLabs/Meadow.Project.Samples/blob/main/Source/Hackster/MeadowClock/MeadowApp.cs#L44-L58) using `Device.PlatformOS.SetClock`.
closed 2023-10-02 by patridge