WildernessLabs/Documentation #306: Getting Started - Hello Meadow instructions incorrect for Windows Visual Studio

rcdemski, 2022-04-09:
The Getting Started guide topic "[Hello, Meadow](http://developer.wildernesslabs.co/Meadow/Getting_Started/Hello_World/)" Step 3 instructions for Windows are out of date with the new v0.19.0 Visual Studio extensions. 
Instead of referencing opening the "Meadow Devices" window, they should reference adding the "Meadow Device Panel" toolbar as discussed in the [v6.2 announcement blog post](https://blog.wildernesslabs.co/b6-2-is-live-with-new-meadow-os-improvements-and-m-f-drivers/) under the topic "Visual Studio (for Windows) Extension"

mongeon, 2022-05-06:
@jorgedevs You can close this issue, since the changes have been merged :)

Thanks.

jorgedevs, 2022-05-06:
Oh yes, thanks for the contribution @mongeon ! 👍🏼

closed 2022-05-06 by jorgedevs