WildernessLabs/Documentation #338: Getting Started - .NET Tools path needs to be added on mac

bryancostanich, 2022-07-20:
can be done this way:
`export PATH="$PATH:/Users/<username>/.dotnet/tools"`
Add it to ~/.zshrc

but we need to figure out that magic shell incantation so users can just run a couple of commands.

after installing brew, it has you do this, but i couldn't get it to work with the dotnet tools path:
```bash
echo 'eval "$(/opt/homebrew/bin/brew shellenv)"' >> /Users/bryancostanich/.zprofile
eval "$(/opt/homebrew/bin/brew shellenv)"
```

bryancostanich, 2022-07-20:
also, seems to be `.zprofile` on my computer, not `.zshrc`

patridge, 2022-09-02:
I think we have this resolved in the beta docs (in the `develop` branch here), here with an `export` call.
http://beta-developer.wildernesslabs.co/Meadow/Getting_Started/Deploying_Meadow/#download-meadow-os-and-network-binaries-1

patridge, 2022-11-04:
This should be deployed to production with the RC-1 docs publish.

closed 2022-11-04 by patridge