WildernessLabs/Documentation #382: LED API docs only offer LedBarGraph sample

patridge, 2022-11-07:
Pages:
* http://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Leds.Led.html
* http://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Leds.PwmLed.html

These [docs for both LED](http://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Leds.Led.html) and [PWM LED](http://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Leds.PwmLed.html) only offer docs specific to the `LedBarGraph` class. It would probably be nice to include a much simpler LED and PWM LED example.

Besides not showing the expected example, it can cause further confusion. Coming from the [Peripherals page](http://developer.wildernesslabs.co/Meadow/Meadow.Foundation/Peripherals/), it shows 
links for `Led` and `LedBarGraph`. But when going to `Led` and only seeing an example for `LedBarGraph`, it makes you think you clicked the wrong link.

NevynUK, 2023-12-07:
Links to the sources also take you to the LED bar graph samples in github.

patridge, 2024-06-16:
Looks like we have a mixed update. Those linked pages no longer reference LedBarGraph, but there are now new very-minimally different URLs for those two API endpoints that _do_ still reference only LedBarGraph. We now want to address this issue on these two pages. I'll update the initial report accordingly.

* https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Leds/PwmLed/
* https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Leds.LedBarGraph/

patridge, 2024-06-16:
@adrianstevens Is it possible that the code snippet system is incorrectly identifying the `Meadow.Foundation.Leds.LedBarGraph` sample for the plain `Meadow.Foundation.Leds` page instead of this one?

https://github.com/WildernessLabs/Meadow.Foundation/blob/develop/Source/Meadow.Foundation.Core.Samples/Leds.Led_Sample/MeadowApp.cs