WildernessLabs/Documentation #394: Windows deployment docs unclear about bootloader rmode

KallDrexx, 2022-12-03:
First time meadow user who just got a project lab, and the docs were way more confusing than they should be.  The http://developer.wildernesslabs.co/Meadow/Getting_Started/Deploying_Meadow/ URL is the URL  I get pointed to when visiting https://wldrn.es/ProjectLabStart and click [Setup your Meadow Build Environment](http://developer.wildernesslabs.co/Meadow/Getting_Started/Deploying_Meadow/).

The "Update USB driver for ST devices" section states:

> Connect a Meadow device in bootloader mode

Long story short, at this point new users have no idea what this means nor how to get into bootloader mode.  It took me significant time to realize that the reason zadig wasn't showing it was because I wasn't actually in boot loader rmode, and it wasn't until I accidentally scrolled down to see a "Put the device into DFU Bootloader mode." section below that actually explained how to put my meadow into bootloader mode.  

It's confusing not putting the bootloader mode above the OS sections.

patridge, 2023-10-13:
It looks like the evolution of the Windows OS section around dealing with USB drivers caused most of this issue. We have a section on putting Meadow into bootloader mode, but the Windows Zadig section is before that.

We probably just need to move this section...
https://github.com/WildernessLabs/Documentation/blob/main/docs/Meadow/Getting_Started/Deploying_Meadow/index.md?plain=1#L77-L84

Up above this section...
https://github.com/WildernessLabs/Documentation/blob/main/docs/Meadow/Getting_Started/Deploying_Meadow/index.md?plain=1#L42-L68

...since that's a prereq for things inside the DFU stuff.

AlexHedley, 2023-11-30:
@patridge good to close too.

patridge, 2024-06-16:
Yep, sorry for the delay...looks resolved.

closed 2024-06-16 by patridge