WildernessLabs/Documentation #400: Remove old setup page

adrianstevens, 2022-12-16:
http://developer.wildernesslabs.co/Meadow/Getting_Started/Setup/

patridge, 2023-10-02:
It looks like the macOS and Windows sub-pages are now just stubs to redirect back to the main Getting_Started page.

e.g., http://developer.wildernesslabs.co/Meadow/Getting_Started/Setup/macos/

@adrianstevens Do we want to continue that effort on this page?

patridge, 2024-06-13:
Setup is still showing in the left-hand nav, though it is mostly redirecting now.

Possible approach:

* [ ] Remove left TOC **Setup** section
* [ ] Rework the root Setup page to point directly to the new pages rather than linking to pages that point to the new location
* [ ] Either...
    * [ ] Leave old setup pages with new location for anyone with old links
    * [ ] Add redirects to send folks to new locations directly

patridge, 2024-06-16:
It looks like the TOC is just automatically happening when you load a page that isn't part of the usual sidebar system. So, removing that checklist item.

closed 2024-06-24 by patridge