WildernessLabs/Documentation #402: "Hello World" does not work on macOS

loicmorvan, 2022-12-19:
I followed the [Hello World example](http://developer.wildernesslabs.co/Meadow/Getting_Started/Hello_World/) on my Mac, but did not succeed.
I have installed Visual Studio 2022 for Mac Community, then installed the extension called "Meadow for VS2022".
At this installation step, VS tells me :
[image]

I have verified that my install of VS2022 includes Xamarin: the documentation
 talks about "Mobile Development with .NET" but this option does not exist in VS2022 for Mac, instead I have the Android checked in the Xamarin section:
[image]

At this point, I'm not able to create a Meadow Application project: it does not appear in the "New project types".

CartBlanche, 2024-03-04:
@loicmorvan Since Dec 2022 we've released v1..x of our extension. Is this still an issue for you or can we close this issue??

loicmorvan, 2024-03-04:
I did not test it since a while... By the way visual studio for Mac will be discontinued soon, so I don't think this issue is worth it.

CartBlanche, 2024-03-04:
@loicmorvan we are supporting VS4 Mac until August. We already have VSCode support for Mac working, which we'll encourage everyone on Mac to move to. 

I'll close this issue for now and if you still hit it using VS4M just reopen it.

closed 2024-03-04 by CartBlanche