WildernessLabs/Documentation #440: docfx migration

lamebrain, 2023-04-11:
Sadly, our Jenkins server died, RIP. The scripts for generating the docs have been [migrated to actions](https://github.com/WildernessLabs/Documentation/blob/main/.github/workflows/jekyll-beta.yml), but the API docs are sadly outdated. Here's the script from the Jenkins job:
```
cd Documentation
cd docfx
docfx docfx.json
(cat .\docs\api\Meadow\toc.html) | % { $_ -replace ">Meadow.Peripherals.", ">" } > .\docs\api\Meadow\toc.html
(cat .\docs\api\Meadow.Foundation\toc.html) | % { $_ -replace ">Meadow.Foundation.", ">" } > .\docs\api\Meadow.Foundation\toc.html
(cat .\docs\api\Meadow.Foundation\toc.html) | % { $_ -replace ">/a>", ">Meadow.Foundation</a>" } > .\docs\api\Meadow.Foundation\toc.html
cd ../docs
bundle install
bundle exec jekyll build
cp ../docfx/docs ./_site -r
aws s3 sync _site s3://developer.wildernesslabs.co --delete
```
(The jekyll portion isn't needed)

For this to work, Meadow and Meadow.Foudation repos need to be pulled as siblings.

CartBlanche, 2023-04-12:
@lamebrain So is the idea we'll have a api-doc.yml in this repo, that will contain the the docfx command calls?

lamebrain, 2023-04-13:
When there is a change made in Meadow or Meadow.Foundation, we need clone the repo as a sibling, run the `docfx docfx.json` command, and copy the output the developer site.

lamebrain, 2023-04-13:
maybe @patridge could help with this too?

patridge, 2023-10-01:
This was added in #493 and others.

PR #444 (2023-07-22): API docs publishing DocFx CI
Fixes #440

## Risk and recovery

* If something goes horribly wrong and it accidentally wipes out the current beta-develop API docs, we can copy the /api folder from production back over top of beta-develop with the help of someone with AWS access.
* If something goes super-extra horribly wrong and it accidentally wipes out all of beta-develop's docs content entirely somehow, we can run the existing beta docs publish Action to restore regular docs, and then run the above AWS copy back from production.

## Implementation details

* Deployment of API docs will only  [... cut]
closed 2023-10-01 by patridge