WildernessLabs/Documentation #480: HelloPulsy docs inaccurate for F7FeatherV1

davisar, 2023-07-10:
I'm using the F7 Micro V1 and working through some of the docs and found this one to be inaccurate for the v1 microcontroller (potentially also v2): http://developer.wildernesslabs.co/Meadow/Meadow.Foundation/Getting_Started/ 

I'm not sure if it's also out of date for the v2 board, but the issue I ran into was that the PwmLed class doesn't take a "Device" parameter as indicated in the documentation.

Instead here is code that worked on the v1 device:

```
using System.Threading.Tasks;
using Meadow;
using Meadow.Devices;
using Meadow.Foundation.Leds;

namespace HelloPulsy
{
    // Change F7FeatherV2 to F7FeatherV1 for V1.x boards
    public class MeadowApp : App<F7FeatherV1>
    {
        PwmLed pwmLed;

        public override Task Initialize()
        {
            //new PwmLed()
            pwmLed = new PwmLed(Device.Pins.D13, TypicalForwardVoltage.Blue);
            return base.Initialize();
        }

        public override Task Run()
        {
            // pulse the LED
            pwmLed.StartPulse();
            return base.Run();
        }
    }
}
```

patridge, 2023-10-12:
This appears to be legacy from when we required the `Device` parameter. That's now handled for us, so we just need to remove that parameter entirely.

engunneer, 2023-11-30:
this should be able to be closed @adrianstevens ?

patridge, 2023-11-30:
Yep...thanks for the fix, @AlexHedley!
And thanks for the heads-up, @engunneer.

closed 2023-11-30 by patridge