WildernessLabs/Documentation #485: Hackhit Blog Reference Issues - Burning out LEDs

davisar, 2023-07-16:
This blog appears to have an issue with burning out certain LEDs: https://blog.wildernesslabs.co/hack-kit-series-closer-look-at-relays/

The blog has you drive the relay to pass through a 3.3v load from the Meadow device. This works well for the Blue LED, which can handle a 3.3v load, however the RED LED does not appear to support this load and burns out.

The diagram and picture in the blogpost also do not match - the wiring configuration differs between the picture and diagram for relay#2 (the relay controlling the blue LED).

Ran into this issue when using the Meadow HackKit with the F7 Micro v1 board.

engunneer, 2023-11-30:
totally agree on this needing a fix. 'blinking an LED' tutorials are great for introducing current limiting resistors.

patridge, 2024-06-16:
@jorgedevs Do we have the original Fritzing diagrams that we can update to reference some resistors when we update the text accordingly?

patridge, 2024-06-24:
Once we have a chart for #83, it might be worth referencing for any solution here.