WildernessLabs/Documentation #549: More warnings about charge-only USB cables

patridge, 2023-10-12:
There is a warning about charge-only (no-data) USB cables in the [Troubleshooting page](https://developer.wildernesslabs.co/Meadow/Meadow_Basics/Troubleshooting/#usb-cables), but it would be nice to have a version of that on the [Meadow OS Deployment page](https://developer.wildernesslabs.co/Meadow/Getting_Started/Deploying_Meadow/) as well, possibly under each OS when first connecting the Meadow, since that's likely where folks will first hit it.

AlexHedley, 2023-10-15:
Do you want the following added under Sections 1, 2 & 3?

> ## USB Cables
> It is important to remember that not all USB cables are created equal. When programming, Meadow uses a USB cable supplying both power and data. It is important to use a good quality USB cable to connect your Meadow device to your computer.
> Some USB cables are only suitable for charging as they do not contain the D+ and / or D- data lines.
> It is important to ensure that you have a USB cable capable of providing data transfer capability and that it is a good quality cable.

> ## Unpowered USB Hub
> If you’re having issues communicating with Meadow that is connected via an unpowered USB-Hub, try connecting Meadow directly to a USB port in your machine to ensure the USB-Hub is the issue.
> As there are a large variety USB-hubs out there, we cant guarantee Meadow will work properly for all of them.

patridge, 2024-06-16:
Sorry we missed this reply. Yes, that's was the general idea we should summarize on the Deploying page. It can be super frustrating when nothing works simply because your cable is fighting you. I'll put something together.

patridge, 2024-06-16:
[PR submitted](https://github.com/WildernessLabs/Documentation/pull/772)

PR #772 (2024-06-24): Add cable/power warnings
{fixes #549}

Adding a notes section about data or poor-quality cables and potential issues with unpowered USB hubs, similar to the sections from the Troubleshooting page.
closed 2024-06-24 by patridge