WildernessLabs/Documentation #644: cellular.GetSignalQuality() example has error

fredmbarrett, 2024-02-06:
The code sample for "[Fetching Cell Signal Quality](https://beta-developer.wildernesslabs.co/Meadow/Meadow.OS/Cellular/#fetching-cell-signal-quality)" shows a variable "var cell = Device.NetworkAdapters.Primary<ICellNetworkAdapter>()" being assigned, but then the next line of code uses a different variable name:
```
var **cell** = Device.NetworkAdapters.Primary<ICellNetworkAdapter>();
double csq  = **cellAdapter**.GetSignalQuality();
```
Also, assuming the cell adapter's GetSignalQuality() method is called, does that also update the state of the CellAdapter.Csq property?

patridge, 2024-02-24:
I've got a fix in PR #676.

@duduita, do you happen to have an answer for the follow-up question?

duduita, 2024-02-26:
Looking at the code, it seems that the `CellAdapter.Csq` will keep retrieving the cached signal quality obtained during the cellular connection. I should adjust the `GetSignalQuality()` method to update the `_csq` private variable when called, consequently, the `Csq` will retrieve the most updated signal quality fetched.

PR #676 (2024-02-27): Fix variable name mismatch
{fixes #644, thank you @fredmbarrett}
closed 2024-02-27 by patridge