WildernessLabs/Documentation #783: F7 Feather can not be recognized on Surface Pro 8 for some reason

sergiibielskyi, 2024-07-08:
Hey,
I just received hack kit from meadow. I am trying to connect MCU via USB. When I connect such a device to the laptop I see a blue light near BOOT button. I follow the instruction to deploy OS https://developer.wildernesslabs.co/Meadow/Getting_Started/Deploying_Meadow.OS/ I do not see any new STM32 BOOTLOADER or USB Serial Device (COMX) in Device Manager. I changed and used different interfaces. What am I doing wrong? )

jorgedevs, 2024-07-08:
HI Sergii,

If you connect your board just regular (not holding BOOT button). Does a new COM port show up in the Device Manager?

sergiibielskyi, 2024-07-08:
Hey, when I connect device, there is nothing changed in device manager (

closed 2024-07-08 by sergiibielskyi

jorgedevs, 2024-07-09:
If you're still having issues, I suggest you join our [Public Slack](http://slackinvite.wildernesslabs.co/) channel. It'll be easier to help you there directly.